mciepluc/cocotb-coverage #6: CRITICAL cocotb.regression   regression.py:132  in initialise      Failed to import module test_fifo: 'return' with argument inside generator (test_fifo.py line 133)

dywcheng, 2019-01-13:
Hi,

I downloaded the lib and ran make SIM=icarus -d because it failed import test_fifo.py due to Syntax error.  I didn't do any modification.  Here is the long version of the message.  Please help.  Thanks.

    0.00ns INFO     cocotb.regression                         regression.py:133  in initialise                      MODULE variable was "test_fifo"
    0.00ns INFO     cocotb.regression                         regression.py:134  in initialise                      Traceback: 
    0.00ns INFO     cocotb.regression                         regression.py:135  in initialise                      Traceback (most recent call last):
                                                                                                                      File "/home/dywcheng/nonShareVM/cocotb/cocotb/regression.py", line 130, in initialise
                                                                                                                        module = _my_import(module_name)
                                                                                                                      File "/home/dywcheng/nonShareVM/cocotb/cocotb/regression.py", line 65, in _my_import
                                                                                                                        mod = __import__(name)
                                                                                                                    SyntaxError: 'return' with argument inside generator (test_fifo.py, line 133)

mciepluc, 2019-01-13:
@dywcheng 
Are you calling make from cocotb-coverage/examples/fifo/tests dircetory?
Have you updated your path to COCOTB in the Makefile?
Therefore, is your PYTHONPATH correct?

dywcheng, 2019-01-13:
Answers are all YES.  Please see below.

dywcheng@ubuntu:~/nonShareVM/cocotb/cocotb_coverage/examples/fifo/tests$ !ma
make SIM=icarus
make results.xml
make[1]: Entering directory
'/home/dywcheng/nonShareVM/cocotb/cocotb_coverage/examples/fifo/tests'
PYTHONPATH=/home/dywcheng/nonShareVM/cocotb/build/libs/x86_64:/home/dywcheng/nonShareVM/cocotb:/home/dywcheng/nonShareVM/cocotb/cocotb_coverage/examples/fifo/tests:/home/dywcheng/nonShareVM/cocotb/cocotb_coverage/examples/fifo/tests/../../..:
LD_LIBRARY_PATH=/home/dywcheng/nonShareVM/cocotb/build/libs/x86_64::/usr/lib:/usr/lib:/usr/lib:/usr/lib
MODULE=test_fifo \
        TESTCASE= TOPLEVEL=fifo_mem TOPLEVEL_LANG=verilog \
        /usr/bin/vvp -M /home/dywcheng/nonShareVM/cocotb/build/libs/x86_64
-m gpivpi sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi
gpi_embed.c:111  in embed_init_python               Did not detect Python
virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi
GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
    0.00ns INFO     cocotb.gpi
gpi_embed.c:278  in embed_sim_init                  Running on Icarus
Verilog version 0.9.7
    0.00ns INFO     cocotb.gpi
gpi_embed.c:279  in embed_sim_init                  Python interpreter
initialised and cocotb loaded!
    0.00ns INFO     cocotb
__init__.py:132  in _initialise_testbench           Running tests with
Cocotb v1.0.1 from /home/dywcheng/nonShareVM/cocotb
    0.00ns INFO     cocotb
__init__.py:148  in _initialise_testbench           Seeding Python random
module with 1547371833
    0.00ns CRITICAL cocotb.regression
 regression.py:132  in initialise                      Failed to import
module test_fifo: 'return' with argument inside generator (test_fifo.py,
line 133)
    0.00ns INFO     cocotb.regression
 regression.py:133  in initialise                      MODULE variable was
"test_fifo"
    0.00ns INFO     cocotb.regression
 regression.py:134  in initialise                      Traceback:
    0.00ns INFO     cocotb.regression
 regression.py:135  in initialise                      Traceback (most
recent call last):

                                          File
"/home/dywcheng/nonShareVM/cocotb/cocotb/regression.py", line 130, in
initialise

                                            module = _my_import(module_name)

                                          File
"/home/dywcheng/nonShareVM/cocotb/cocotb/regression.py", line 65, in
_my_import

                                            mod = __import__(name)

                                        SyntaxError: 'return' with argument
inside generator (test_fifo.py, line 133)

Traceback (most recent call last):
  File "/home/dywcheng/nonShareVM/cocotb/cocotb/__init__.py", line 168, in
_initialise_testbench
    regression.initialise()
  File "/home/dywcheng/nonShareVM/cocotb/cocotb/regression.py", line 130,
in initialise
    module = _my_import(module_name)
  File "/home/dywcheng/nonShareVM/cocotb/cocotb/regression.py", line 65, in
_my_import
    mod = __import__(name)
  File
"/home/dywcheng/nonShareVM/cocotb/cocotb_coverage/examples/fifo/tests/test_fifo.py",
line 133
    return data, success
SyntaxError: 'return' with argument inside generator
Cocotb initialisation failed - exiting
    0.00ns ERROR    cocotb.scheduler
__init__.py:191  in _sim_event                      Failing test at
simulator request before test run completion: Simulator shutdown prematurely
make[1]: Leaving directory
'/home/dywcheng/nonShareVM/cocotb/cocotb_coverage/examples/fifo/tests'

Are you calling make from cocotb-coverage/examples/fifo/tests dircetory?

mciepluc, 2019-01-13:
Looks like there is an issue with using ```return``` in function ```process_data```. As it is a generator (coroutine) it shoud return value by ```yield```. According to [explanation](https://stackoverflow.com/questions/16780002/return-in-generator-together-with-yield-in-python-3-3) current implementation is acceptable with Python 3.3 and above. I will take a look and possibly fix it.

@dywcheng please note you need to use Python 3 anyway for this package, so if you switch to the recent Python distribution, this issue should not be present.

mciepluc, 2019-01-14:
@dywcheng it depends on your system setup. You should play with your environment variables (```PATH```, ```PYTHONPATH```, ```PYTHONHOME```) to make sure cocotb runs on Python 3. You can try setting ```PYTHON_BIN=python3```. If you need further help on that you can ask within cocotb project.

jahagirdar, 2019-05-15:
Came to this thread because I had the same problem.
Leaving the solution here in case I forget it and come again next year :)

```
python3.6 -m venv `pwd`
source /bin/activate
make
```