mciepluc/cocotb-coverage #18: Run on Azure with coverage report ?

themperek, 2019-05-14:
Would it work if we setup CI with Azure pipelines and report coverage directly like here: https://docs.microsoft.com/en-us/azure/devops/pipelines/test/review-code-coverage-results?view=azure-devops
I can try to set this up.

mciepluc, 2019-05-22:
Hi @themperek I am not sure if it is not too much work to be done and what will be the benefit. This package is not expected to grow fast and current test suite seems to be quite a reasonable verification approach. If code coverage can be easily setup without too much additional complexity that's fine, but in other case I would skip it for now.

mballance, 2020-04-07:
@themperek, I'm curious if you've found a way to report functional coverage-like results on Azure pipelines. I looked at trying to force-fit this type of data into one of the supported code-coverage formats, but it didn't look very promising...

themperek, 2020-04-07:
I started to fit into Cobertura line coverage (it possible with a bit of hacking) but I think in the end we need to generate our own HTML report that can be displayed.  Check this also: https://github.com/mciepluc/cocotb-coverage/issues/29