mciepluc/cocotb-coverage #24: Documentation improvements

mciepluc, 2019-08-21:
As this project is mainly addressed for SystemVerilog and commercial tools users to encourage them to abandon the sinking ship, it is good to provide a more informative examples and tutorials. The goal is to show consistency with SystemVerilog coverage/randomization features with much higher flexibility and user-friendliness. 
Also @cmarqu provided an interesting tutorial: https://www.divio.com/blog/documentation/
Please share any suggestions what should be included in the documentation.
Some suggestions:
- FIFO example documentation
- add more examples, focus on CRV (simple protocol, simple packet router?)
- tutorial how to translate SV syntax to cocotb-coverage
- tutorial what you can't do (easily) in SV and can do in cocotb-coverage

cmarqu, 2019-08-21:
A table that shows the correspondence between SV and cocotb-coverage syntax would be helpful. Bonus points if it fits onto a single page to be used as a reference card (reminds me, we also need this for cocotb).

Examples... the simple packet router seems to be a default design for teaching UVM, so we can't go wrong with this.
Also, Lars Asplund of VUnit has started a big undertaking of comparing VUnit to SV-UVM (in https://gitter.im/VUnit/vunit_vs_uvm) by using the tutorial at http://cluelogic.com. This page may give some more ideas.

pwiecha, 2019-08-22:
Hi, as you both know I've started the yas_router some time ago but didn't have time to work on it until now. I'm planning to implement CRV at some point alongside sphinx documentation, maybe some articles about building a testbench in cocotb as well. This is my contribution proposal.

mciepluc, 2019-08-22:
@pwiecha that's ok to prepare a more complex example as an external project. For a simple demo here (like already existing FIFO) I think that simple packet switch should be enough.

mciepluc, 2020-02-26:
Updated documentation is committed, still a few tutorials needs to be done. I hope that this level of documentation should be sufficient for everyone.

mciepluc, 2020-06-16:
Dear All, I finished writing the tutorials and the documentation is pretty much completed now. I encourage you to review it and share your comments. @cmarqu @pwiecha, other users, your comments will be very welcome!
https://cocotb-coverage.readthedocs.io/en/latest/

mciepluc, 2020-08-07:
documentation reviewed

closed 2020-08-07 by mciepluc