mciepluc/cocotb-coverage #27: Ignore whitespace in signal names

sjalloq, 2019-10-04:
Hi,

I just found a very obscure issue caused by me specifying the add_rand() method with whitespace at the end of the signal names.  I was using rectangle edit in emacs to insert a bunch of long names and ended up with something like:

```
self.add_rand("sig_name_a  ", list(range(10)))
self.add_rand("sig_name_ab", list(range(10)))
```

which when you run a test, ends up with 0 always assigned to the signals with whitespace.  In the example above, sig_name_a is never randomized.

Could you fire a warning/error for this or just strip off any whitespace?

Cheers.

mciepluc, 2019-10-07:
Hi @sjalloq
The string in the ```add_rand``` method must be equal to the variable name, so should not contain any whitespaces. But you are right, there should be a warning fired when a random variable does not exist. 2B added.

mciepluc, 2020-01-09:
added exception when trying to add_rand of the non-existing class member

closed 2020-01-09 by mciepluc