mciepluc/cocotb-coverage #31: Error in merging two or more files

Losiek, 2020-01-21:
Following error is occurs when merging two or more files with function 
`cov.merge_coverage(print, "cov.xml", "cov_1.xml", "cov_2.xml")`

Error:
> ~/.conda/envs/p_env/lib/python3.7/site-packages/cocotb_coverage/coverage.py in merge_element(db)
   1082                     parent_hits_threshold = int(
   1083                         name_to_elem[parent_name].attrib['at_least'])
-> 1084                     if (hits_orig < parent_hits_threshold
   1085                         and hits_orig+hits >= parent_hits_threshold):
   1086                         update_parent(name=abs_name, bin_update=True,
> UnboundLocalError: local variable 'hits_orig' referenced before assignment

mciepluc, 2020-01-22:
@Losiek please confirm it's fixed and close

Losiek, 2020-02-04:
Works, thanks.

closed 2020-02-04 by Losiek