mciepluc/cocotb-coverage #37: randomize() function generates repeated values

Divyanirankari, 2020-02-12:
How to generate non repetitive values using randomize() function in crv???

mciepluc, 2020-02-19:
@Divyanirankari 
I am not sure what do you mean. Do you mean to make sure variables are different? In that case you should define a constraint (e.g. for variables ```x``` and ```y```):
```Python
add_constraint(lambda x, y: x != y)
```
Or you mean that each randomization should guarantee non-repetitive results? 
In that case you should define a set of already used values and define constraint which prevenrs from picking values from this set. Example for variable ```x```:
```Python
class Foo(Randomized):
    __init__(self):
    self.used = []
    self.x = 0
    self.add_rand("x", range(100))

    def avoid_used(x):
        return x not in used

    self.add_constraint(avoid_used)
...
object = Foo()
object.randomize()
object.used.add(object.x)
```

Divyanirankari, 2020-02-19:
@mciepluc 
Thanks.
i want  each randomization should guarantee non-repetitive results.

closed 2020-05-21 by mciepluc