mciepluc/cocotb-coverage #39: constrained random and functional coverage

Divyanirankari, 2020-04-24:
```
hi, 
  i have to randomise  some bits of 32 bit instruction 
  and some bits will be constant like below format 
  
+---------+-------+-------+-----+-------+---------+
| 0100000 | XXXXX | XXXXX | 111 | XXXXX | 0110011 |
+---------+-------+-------+-----+-------+---------+

through direct random function we can randomise cant find coverage
 using cocotb-coverage library.

    for i in range(n):
            instruction = random.randint(0,0xFFFFFFFF)
            instruction = instruction & 0x01FF8F80
            
            instruction = instruction | 0x40007033
            
            
            
how we can find constrained random and coverage using cocotb-coverage?

```

mciepluc, 2020-04-25:
@Divyanirankari 
You should NOT use constraint solver for such a task. This is a typical scenario of usage of the random number generator, as you provided. Please use ```random.randint``` or ```np.random.randint```. In order to combine random generation with contraint solver, please use ```pre_randomize()``` or ```post_randomize()``` functions. 

Please look at the correct usage in the provided [example](https://cocotb-coverage.readthedocs.io/en/latest/examples.html#id2). Note that randomized are fields that have small domains, so constraint solver can easily find the solution. For larger domain - generating the packet payload - a ```post_randomize()``` and ```random.randint``` is used.

Divyanirankari, 2020-04-25:
@mciepluc 
Ok will use pre_randomize() and post_randomize().
How to find coverage for this type of scenario.?

Divyanirankari, 2020-04-25:
What will be the bins for this scenario in Coverpoint. ?. I am not able to understand?

mciepluc, 2020-04-27:
@Divyanirankari 
I am sorry but I do not get your question. You as a designer should know what should be included in the functional coverage scope and define the bins accordingly.

Divyanirankari, 2020-04-27:
@mciepluc 
Ok Thanks

closed 2020-04-27 by Divyanirankari