mciepluc/cocotb-coverage #40: solve_order issue

tsengr0916, 2020-05-16:
I have the following test code. For some reason, if I specify the solver_order statement, it will always give me an error:
```
class my_random(Randomized):  

  def __init__(self):
    super().__init__()    
    self.x = 0
    self.y = 0
    self.z = 0  

    self.add_rand('x', list(range(10)))
    self.add_rand('y', list(range(10)))
    self.add_rand('z', list(range(10)))

    self.add_constraint(lambda x: 0 <= x <= 5)
    self.add_constraint(lambda y: 0 <= y <= 6)
    self.add_constraint(lambda x,y,z: x+y+z < 15)    

    self.solve_order(['x', 'y'], 'z')

  def print_vars(self):
    print("x={}, y={}, z={}, sum={}".format(self.x, self.y, self.z, self.x+self.y+self.z))   

if __name__ == '__main__':
  my_obj = my_random()
  my_obj.add_constraint(lambda x,y,z: x+y+z < 10)
  my_obj.randomize()
  my_obj.print_vars()
```

The error is like this:
File "/prj/tbs/5g/sandiego/hw/scratch/sims/hctseng/pandace3/venv3.5.2/lib/python3.5/site-packages/cocotb_coverage/crv.py", line 401, in _del_constraint
    assert(0), "Could not delete a constraint!"
AssertionError: Could not delete a constraint!

If I change the solve_order() in this way, it will give me the same error:
self.solve_order('x', 'y', 'z')

Other questions:
If I have multiple solve_order() statements, will it honor all of them, or just the last statement?

What is the definition of hard and soft constraint in COCOTB? 

Many thanks!
Richard

mciepluc, 2020-05-20:
@tsengr0916 
Just tried to run your example and it seems working... Have you installed cocotb-coverage from pip? Can you install from github directly and check?
Regarding ```solve_order```, the later call is overwriting the existing ones, so you basically should not specify this function multiple times. I will add a description in doc.

tsengr0916, 2020-05-21:
Thanks. Yes I did pip install for cocotb, python-constraint, pyyaml, and cocotb-coverage. Here is the list of versions:

cocotb (1.4.0.dev0)
cocotb-coverage (1.0.0)
pip (8.1.1)
python-constraint (1.3.1)
PyYAML (5.3.1)
setuptools (20.10.1)

Let me know if I missed something.

Many thanks!

tsengr0916, 2020-05-21:
BTW, I noticed that if I put all variables in one list, it will work:
self.solve_order(['x', 'y', 'z'])
 
I'm using Python version 3.5.2.

Thanks!

mciepluc, 2020-05-21:
@tsengr0916 
Please install from github directly, the official 1.0.0 release does not contain several updates:
```pip install -U git+https://github.com/mciepluc/cocotb-coverage@master```

tsengr0916, 2020-05-22:
@mciepluc
I tied the pip install command you provided, but got the following error:

Collecting git+https://github.com/mciepluc/cocotb-coverage@master
  Cloning https://github.com/mciepluc/cocotb-coverage (to master) to /tmp/pip-76f4ouvn-build
fatal: unable to access 'https://github.com/mciepluc/cocotb-coverage/': error:140770FC:SSL routines:SSL23_GET_SERVER_HELLO:unknown protocol
Command "git clone -q https://github.com/mciepluc/cocotb-coverage /tmp/pip-76f4ouvn-build" failed with error code 128 in None

Did I miss something else in my environment?

Thanks

mciepluc, 2020-05-24:
@tsengr0916 I don't know what this error is about. It is related to your setup, not to cocotb-coverage.

tsengr0916, 2020-05-25:
@mciepluc. I manually download it to my project space and tried it again. It works now. I will follow up the installation issue with our IT guys. Thanks for your help!

closed 2020-05-25 by tsengr0916