mciepluc/cocotb-coverage #42: Typos in documentation

pwiecha, 2020-05-21:
Documentation reference.html [https://cocotb-coverage.readthedocs.io/en/latest/reference.html](url) **CoverPoint** example:
```python
>>> @coverage.CoverPoint( # cover (arg/2) < 1...5 (5 bins)
...
bins = list(range(1, 5))
```
should be 4 bins

**CoverCross** example:
4 bins total, 4x4 - 2 = 14 bins total

[https://cocotb-coverage.readthedocs.io/en/latest/examples.html](url)
**Testbench section**

`bins = list(range(3,32))                                #may be 3 ... 32 bytes`
Should be 31 bytes, range() is not inclusive

And below that in the TB code:
```python
    low_limit = random.randint(3,31)
    up_limit = random.randint(low_limit,32)
````

Here I would suggest using `randrange(3, 32)` and `randrange(low_limit, 33)` because it is consistent with range() and how indexing works in python and the rest of documentation.

pwiecha, 2020-05-21:
I can do that if you approve of these changes.

mciepluc, 2020-05-21:
fixed

closed 2020-05-21 by mciepluc