mciepluc/cocotb-coverage #43: Example FIFO gets unexpected output

AIAFammy, 2020-05-22:
Hi, I have installed cocotb, cocotb-coverage, iverilog, and icarus successfully, and also the fifo example can run without error by "make SIM=icarus".
But the result seems to be something wrong. 

The output below run by 10 random operations. 
I expect that "read" after "write" should read the data in fifo successfully, but it always shows "Data NOT read, fifo EMPTY!".

`MODULE=test_fifo TESTCASE= TOPLEVEL=fifo_mem TOPLEVEL_LANG=verilog \
        /usr/bin/vvp -M /home/user/.conda/envs/py36/lib/python3.6/site-packages/cocotb-1.4.0.dev0-py3.6-linux-x86_64.egg/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:71   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:106  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:264  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:146  in _initialise_testbench           Running on Icarus Verilog version 0.9.7
     0.00ns INFO     cocotb                                      __init__.py:153  in _initialise_testbench           Running tests with cocotb v1.4.0.dev0 from /home/user/.conda/envs/py36/lib/python3.6/site-packages/cocotb-1.4.0.dev0-py3.6-linux-x86_64.egg/cocotb
     0.00ns INFO     cocotb                                      __init__.py:170  in _initialise_testbench           Seeding Python random module with 1590030689
     0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test test_fifo.fifo_test
     0.00ns INFO     cocotb.regression                         regression.py:453  in _start_test                     Running test 1/1: fifo_test
     0.00ns INFO     ..otb.test.fifo_test.0x2aeb40019080       decorators.py:256  in _advance                        Starting test: "fifo_test"
                                                                                                                     Description:  FIFO Test
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
signal_get: property 18 is unknown
     2.15ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
     2.35ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: 3D
     2.55ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
     2.75ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: 1
     2.95ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
     3.15ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: E
     3.35ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: 66
     3.55ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: AD
     3.75ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
     3.95ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: 42
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                 top : <cocotb_coverage.coverage.CoverItem object at 0x2aeb3fe18da0>, coverage=17, size=32
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.fifo_empty : <cocotb_coverage.coverage.CoverPoint object at 0x2aeb3fe18a90>, coverage=1, size=2
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 10
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN False : 0
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.fifo_full : <cocotb_coverage.coverage.CoverPoint object at 0x2aeb3fe18978>, coverage=1, size=2
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 0
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN False : 10
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.fifo_overflow : <cocotb_coverage.coverage.CoverPoint object at 0x2aeb3fe18ef0>, coverage=1, size=2
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 0
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN False : 10
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.fifo_threshold : <cocotb_coverage.coverage.CoverPoint object at 0x2aeb3fe18e48>, coverage=1, size=2
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 0
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN False : 10
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.fifo_underflow : <cocotb_coverage.coverage.CoverPoint object at 0x2aeb40019cf8>, coverage=1, size=2
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 0
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN False : 10
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.rw : <cocotb_coverage.coverage.CoverPoint object at 0x2aeb3fe18a58>, coverage=2, size=2
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 4
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN False : 6
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.rwXempty : <cocotb_coverage.coverage.CoverCross object at 0x2aeb40019048>, coverage=2, size=4
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN (True, True) : 4
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN (True, False) : 0
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN (False, True) : 6
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN (False, False) : 0
     3.95ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.rwXfull : <cocotb_coverage.coverage.CoverCross object at 0x2aeb40019128>, coverage=2, size=4
     3.95ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN (True, True) : 0
[... cut ...]
                                                                                                                     **                        SIM / REAL TIME : 39.39 NS/S                             **
                                                                                                                     *************************************************************************************

     3.95ns INFO     cocotb.regression                         regression.py:249  in tear_down                       Shutting down...
`

Package versions as below:
   cocotb                 1.4.0.dev0
   cocotb-coverage        1.0.0

Many thanks for your help!

mciepluc, 2020-05-22:
@AIAFammy that looks pretty strange indeed...
A few possible actions that may help to find the issue:
- which version of Icarus do you have? Can you try installing new version (I have a bare installation from sources, but version 10.x should work)
- do you have an option to try a different simulator?
- maybe simulator precision is not enough, can you try modifying clock period from [100 to e.g. 10000](https://github.com/mciepluc/cocotb-coverage/blob/7791d5288bd8c096f822e028622ace44a38394bd/examples/fifo/tests/test_fifo.py#L93)
- anyway please note that official cocotb release is 1.3.1, so maybe something is broken because of that... (however it works on my side)

AIAFammy, 2020-05-22:
@mciepluc Thanks for your answer.

I install Icarus verilog by `sudo apt-get install verilog`, and the version is 0.9.7

First, I install the official cocotb version 1.3.1, and I modified the clock period from 100 to 1000 and 10000. But the output still not correct.
To make sure the clock period is modified correctly, the line I modified is:
https://github.com/mciepluc/cocotb-coverage/blob/master/examples/fifo/tests/test_fifo.py#L93

And then I install Icarus Verilog version 10.3 by clone the source. But the FIFO example gets the following output:

```
make results.xml
make[1]: Entering directory `/home/user/cocotb-coverage-master/examples/fifo/tests'
PYTHONPATH=/home/user/cocotb-coverage-master/examples/fifo/tests/build/libs/x86_64:/home/user/cocotb-coverage-master/examples/fifo/tests:/home/user/.conda/envs/py36/lib/python3.6/site-packages: LD_LIBRARY_PATH=/home/user/cocotb-coverage-master/examples/fifo/tests/build/libs/x86_64::/home/user/.conda/envs/py36/lib:/home/user/.conda/envs/py36/lib:/home/user/.conda/envs/py36/lib:/home/user/.conda/envs/py36/lib MODULE=test_fifo \
        TESTCASE= TOPLEVEL=fifo_mem TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /usr/local/bin/vvp -M /home/user/cocotb-coverage-master/examples/fifo/tests/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
Error: VVP input file 9.7 can not be run with run time version 10.3 (stable)
make[1]: *** [results.xml] Error 1
make[1]: Leaving directory `/home/user/cocotb-coverage-master/examples/fifo/tests'
make: *** [sim] Error 2
```

Error: VVP input file 9.7 can not be run with run time version 10.3 (stable)
I can't find any information about this error, it seems like something cannot support Icarus 10.3.

Thanks.

mciepluc, 2020-05-22:
@AIAFammy you have probably not cleaned your directory that still contains a build done by iverilog 0.9.7

AIAFammy, 2020-05-22:
@mciepluc Thanks for your instant response.

I sudo removed the 0.9.7 iverilog, and still got the same error. Then I found that the problem is `cocotb-coverage/examples/fifo/tests/sim_build`.

After removed the old build and make, it seems to be work very well!

```
make results.xml
make[1]: Entering directory `/home/user/cocotb-coverage-master/examples/fifo/tests'
mkdir -p sim_build
/usr/local/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s fifo_mem -g2012   /home/user/cocotb-coverage-master/examples/fifo/tests/../hdl/fifo.v
/home/user/cocotb-coverage-master/examples/fifo/tests/../hdl/fifo.v:46: warning: extra digits given for sized binary constant.
/home/user/cocotb-coverage-master/examples/fifo/tests/../hdl/fifo.v:48: warning: extra digits given for sized binary constant.
/home/user/cocotb-coverage-master/examples/fifo/tests/../hdl/fifo.v:107: warning: extra digits given for sized binary constant.
/home/user/cocotb-coverage-master/examples/fifo/tests/../hdl/fifo.v:109: warning: extra digits given for sized binary constant.
PYTHONPATH=/home/user/cocotb-coverage-master/examples/fifo/tests/build/libs/x86_64:/home/user/cocotb-coverage-master/examples/fifo/tests:/home/user/.conda/envs/py36/lib/python3.6/site-packages: LD_LIBRARY_PATH=/home/user/cocotb-coverage-master/examples/fifo/tests/build/libs/x86_64::/home/user/.conda/envs/py36/lib:/home/user/.conda/envs/py36/lib:/home/user/.conda/envs/py36/lib:/home/user/.conda/envs/py36/lib MODULE=test_fifo \
        TESTCASE= TOPLEVEL=fifo_mem TOPLEVEL_LANG=verilog COCOTB_SIM=1 \
        /usr/local/bin/vvp -M /home/user/cocotb-coverage-master/examples/fifo/tests/build/libs/x86_64 -m gpivpi sim_build/sim.vvp
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Icarus Verilog version 10.3 (stable)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.3.1 from /home/user/.conda/envs/py36/lib/python3.6/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1590153908
     0.00ns INFO     cocotb.regression                         regression.py:190  in initialise                      Found test test_fifo.fifo_test
     0.00ns INFO     cocotb.regression                         regression.py:348  in execute                         Running test 1/1: fifo_test
     0.00ns INFO     ..otb.test.fifo_test.0x2b7829f061d0       decorators.py:253  in _advance                        Starting test: "fifo_test"
                                                                                                                     Description:  FIFO Test
2150000000000.00ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: C3
2350000000000.00ns INFO     cocotb.test                                test_fifo.py:146  in fifo_test                       Data read from fifo: C3
2550000000000.00ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
2750000000000.00ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: D7
2950000000000.00ns INFO     cocotb.test                                test_fifo.py:146  in fifo_test                       Data read from fifo: D7
3150000000000.00ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
3350000000000.00ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
3550000000000.00ns INFO     cocotb.test                                test_fifo.py:153  in fifo_test                       Data written to fifo: 46
3750000000000.00ns INFO     cocotb.test                                test_fifo.py:146  in fifo_test                       Data read from fifo: 46
3950000000000.00ns INFO     cocotb.test                                test_fifo.py:148  in fifo_test                       Data NOT read, fifo EMPTY!
3950000000000.00ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                 top : <cocotb_coverage.coverage.CoverItem object at 0x2b782a0fa2e8>, coverage=22, size=32
3950000000000.00ns INFO     cocotb.test                                 coverage.py:95   in report_coverage                    top.fifo_empty : <cocotb_coverage.coverage.CoverPoint object at 0x2b782a0fa320>, coverage=2, size=2
3950000000000.00ns INFO     cocotb.test                                 coverage.py:100  in report_coverage                       BIN True : 7
[... cut ...]
                                                                                                                            **                        SIM / REAL TIME : 39942767661478.09 NS/S                 **
                                                                                                                            *************************************************************************************

3951000000000.00ns INFO     cocotb.regression                         regression.py:222  in tear_down                       Shutting down...
```

Thanks for your help!

closed 2020-05-22 by AIAFammy