mciepluc/cocotb-coverage #44: Version 1.1.dev in pypi

tsengr0916, 2020-05-26:
Is this 1.1.dev version available from https://pypi.org/project/cocotb-coverage? If not, can you please add in there?

Thanks,
Richard

themperek, 2020-05-27:
If @mciepluc agree I can upload `1.1.dev0`

mciepluc, 2020-05-27:
@tsengr0916 yeah I think it is a good idea anyway to make the pre-relase. I uploaded the package and description on the main page to use ```1.1.dev0``` as recommended.

closed 2020-05-27 by mciepluc

tsengr0916, 2020-05-27:
Thanks guys. I see this is updated: https://github.com/mciepluc/cocotb-coverage, but here is not: https://pypi.org/project/cocotb-coverage/

What is the process to release to the pypi ? If the former link used for development testing?

mciepluc, 2020-05-28:
@tsengr0916 `1.1.dev0` is not a stable release, so it will not be present under the link you provided. It is here https://pypi.org/project/cocotb-coverage/1.1.dev0/