mciepluc/cocotb-coverage #45: Question about Packet Switch Example

AIAFammy, 2020-05-27:
Hi, I have run this example successfully and almost understand every part of the codes. But there is two points that I can't figure out.

First: Why coverpoint `top.filt_addr` only have 32 possible options?
Since `addr & mask` has 8 binary bits, it should be 256 possible options. Also, I found that was't every packet operation output the `top.filt_addr` bin. There must be some point I missed. I guess this coverpoint bin only activate under some condition and so there is only 32 possible options, but I can't figure out.

https://github.com/mciepluc/cocotb-coverage/blob/master/examples/pkt_switch/tests/test_pkt_switch.py#L187 
```
@CoverPoint(
      "top.filt_addr",  
      xf = lambda pkt, event, addr, mask, ll, ul: addr & mask,  #filtering based on a particular bits in header 
      bins = list(range(32))                                    #all options possible
    )
```

Second: I was wondering why we limited `low_limit` value in 3 ~ 30. Does it mean anything wrong when `low_limit = 31` ?
On the other hand, `range(3, 31)` has values from 3 ~ 30, but `random.randint(3,31)` has values from 3 ~ 31. Am I correct?

https://github.com/mciepluc/cocotb-coverage/blob/master/examples/pkt_switch/tests/test_pkt_switch.py#L197
```
@CoverPoint(
      "top.filt_len_ll", 
      vname = "ll",                    #lower limit of packet length
      bins = list(range(3,31)) 
    )
    @CoverPoint(
      "top.filt_len_ul", 
      vname = "ll",                    #upper limit of packet length
      bins = list(range(3,32)) 
    )
```
https://github.com/mciepluc/cocotb-coverage/blob/master/examples/pkt_switch/tests/test_pkt_switch.py#L243
```
low_limit = random.randint(3,31)
up_limit = random.randint(low_limit,32) 
```

Many thanks for your contributions and helps!

mciepluc, 2020-05-28:
@AIAFammy 

1. The `"top.filt_addr"` coverpoint has 32 bins because it was defined that way. I agree in total there is 256 possibilities, but why coverpoint should cover all of them? The designer decides what should be covered, instead of sampling everything what is possible. This is just an example, it can be implemented different way. I will update comment in the code to make it more clear.
2. The problem is as you described. The `random.randint` ranges are not consistent with `range` keyword. I will change to `np.random.randint` as it is safer and more readable.

AIAFammy, 2020-05-28:
Hi, @mciepluc 
Thanks for your answer. It's pretty clear, and I still one

mciepluc, 2020-05-28:
@AIAFammy I have updated the example that should address all the issues you had. Let me know if anything remains unclear.

AIAFammy, 2020-05-28:
@mciepluc Awesome!

I have browsed all the diff parts, and it's super clear! 
Many thanks!

closed 2020-05-28 by AIAFammy