mciepluc/cocotb-coverage #48: Use iterable range without conversion to list as domain in add_rand examples

mariushegele, 2020-08-25:
The `add_rand` method of the `crv.Randomized` class expects an iterable `domain`. All given examples transform iterable `range`s to `list`s, e.g.:

```
def Point(crv.Randomized):
   def __init__(self, x, y):
        # ...
        self.add_rand('x', list(range(10)))
```

Why is this transformation necessary? It is not scalable to large ranges, because Python will construct the whole list in memory. The use of `domain` within constraint resolving appears to be only iteration. This works on `range`s even more efficiently:

```
$ python3 -m timeit 'for i in range(0xFFFF): pass'
50 loops, best of 5: 8.57 msec per loop
$ python3 -m timeit 'for i in list(range(0xFFFF)): pass'
20 loops, best of 5: 11 msec per loop
```

mariushegele, 2020-08-25:
https://github.com/mciepluc/cocotb-coverage/blob/e4b171189a84b2c14daa619093e7155a6595c852/cocotb_coverage/crv.py#L569 

appears to be a related instance:

```
        # Cartesian product of above
        ducSolutions = list(itertools.product(*ducDomains))

        # merge solutions: constrained ones and all possible distribution 
        # values
        for sol in solutions:
            for ducsol in ducSolutions:
            # ...
```

mciepluc, 2020-08-25:
Hmm, I don't remember just now what was the reason behind. Could be a Python 2-related stuff (which does not matter at the moment) or some internal conversion requirement. Anyway, if you need to create a big domain, it is not recommented to do it using `add_rand`. Please use `pre/post_randomize` instead if you see performance issues. Have a look [here](https://cocotb-coverage.readthedocs.io/en/latest/tutorials.html#randomization-order-and-performance-issues).

mariushegele, 2020-08-28:
I see the point, thank you for the link!

closed 2021-02-11 by mciepluc