mciepluc/cocotb-coverage #49: coverage_section build works only with Verilator and not with Icarus

FrancescoDellanna, 2021-03-09:
Cocotb simulation builds correctly with Verilator and generate a good coverage report, but it fails to build with Icarus.

The build with Icarus succeeds if the coverage_section decorator is removed from the @cocotb.coroutine.

I added comments in the code to help debugging the code (full_adder_tb.py and Makefile).

The provide example is a simple full adder and I am not expecting any RTL or python modelling issues.

[cocotb_build_icarus_8453.zip](https://github.com/mciepluc/cocotb-coverage/files/6109897/cocotb_build_icarus_8453.zip)

mciepluc, 2021-03-10:
@FrancescoDellanna 

In your code you try to call ```stimuli``` function with arguments that are taken from the DUT ports. DUT is however at this point un-initialized, so all the values are 'Z' at this moment. Probably Verilator by default makes a pull down on all the 'Z' ports. 

This code should fail, because you try to sample 'Z' values. Eventually inside the stimuli function you drive the ports, but coverage is sampled already at this moment.

You should either initialize the ports at the very beginning of your test, or (preferred) sample the coverage at the point you already sure the values of the DUT are written.

FrancescoDellanna, 2021-03-10:
Thank you a lot. Your reply was very useful and the code works after I initialize the inputs.

closed 2021-03-10 by mciepluc