mciepluc/cocotb-coverage #64: Problems with tutorial transitions example

jonpovey, 2021-11-03:
The documentation tutorial page section about transitions has several issues:

- bins as list-of-list does not work, throws `TypeError: unhashable type: 'list'` (works if changed to list-of-tuple)
- appending to `addr_prev` in the `rel` function causes the sample to be appended multiple times, once for each bin. The append needs to be moved to a separate `xf` to only happen once
- transition `0 1 2 3` bin cannot be hit as there is a `0 1` bin appearing before it, and `inj` is not set.

Also, `inj` does not work as documented, I had to explicitly set it to `False` to get this example to behave. See #63

mciepluc, 2022-01-15:
A big miss out from my side.

mciepluc, 2022-01-15:
thanks, fixed, don't agree with your last bullet however, check the created test for reference :)

closed 2022-01-15 by mciepluc