mciepluc/cocotb-coverage #78: Exception using randomize_with

shareefj, 2022-07-08:
I'm seeing an exception raised when I try to randomize an object.  I have a class that models an AHB transaction and I'm trying to randomize the number of cycles I stall before driving it on the bus.  So my transaction looks something like:

```python
class Transaction(Randomized):
  def __init__(self, address, etc):
    self.address = address
    etc
    self.delay = 0
    self._delay = "small"
    self.add_rand("_delay", ["none", "small", "large"])

  def post_randomize(self):
    if self._delay == "none":
      self.delay = 0
    elif self._delay == "small":
      self.delay = 5
    etc
```

Then when I try to generate a transaction using:

```python
trxns = [Transaction(addr) for addr in range(10)]
for t in trxns:
  t.randomize_with(lambda _delay: _delay == "none")
```

I get:

```
Traceback (most recent call last):
  File "test_random.py", line 63, in _run
    self._generate_transactions()
  File "test_random.py", line 77, in _generate_transactions
    self._trxns = self._region.generate_transactions(
  File "testbench.py", line 154, in generate_transactions
    t.randomize_with(lambda _delay: _delay == "none")
  File ".venv/lib/python3.9/site-packages/cocotb_coverage/crv.py", line 339, in randomize_with
    raise Exception("Could not resolve implicit constraints!")
Exception: Could not resolve implicit constraints!
```

What am I doing wrong?

Is this library still being developed or are there better alternatives now?  I always seem to hit these issues when I come back to use it.

mciepluc, 2022-07-26:
@shareefj 
Please make sure you use up to date python-constraint solver. Then please attach a full failing example so I can debug.

mciepluc, 2023-11-10:
@shareefj any update on this?

shareefj, 2023-11-10:
Sorry, changed jobs since raising this and can't remember what the issue was.  No way of reproducing now so closing.

closed 2023-11-10 by shareefj