mciepluc/cocotb-coverage #84: Adding constraint to 1x16 tuple makes the memory overflow

wangw18, 2023-06-13:
Hi sir
I want to add some constraint to a 1x16 tuple using "**self.add_rand("a",tup)**" with **tup = [(1,2,3..,16), ()..., ()]**, but if I want to cover all cases, the possibility of tup would be greater than "16!". Is there any other solution to solve it?

mciepluc, 2023-06-13:
Sorry I do not understand your question.

wangw18, 2023-06-14:
Suppose there is a 1x16 tuple tup composed of natural numbers , and the total number of full permutation result of it  is around 10^16. Take these results into **self.add_rand("a",tup)**, as long as the constraints do not conflict, there would be bins that satisfy the constraints, but too many samples would lead to memory overflow. I would like to ask if there is a better solution to this case, could cocotb_coverage generate bins with complex constraints and large scale like sv

mciepluc, 2023-06-14:
First of all, it is not SV that resolves complex constraints - it is a constraint solver embedded in the tool you are using. They may be better or worse solvers.
Usually, good practice is not to use high complexity constraints, please look into the tutorial:
https://cocotb-coverage.readthedocs.io/en/latest/tutorials.html#randomization-order-and-performance-issues

wangw18, 2023-06-14:
Thanks, got it

closed 2023-06-14 by wangw18