mciepluc/cocotb-coverage #86: Switch away from the deprecated cocotb.result.TestFailure

imphil, 2023-11-03:
cocotb 2.0 (currently in the master branch) will remove the deprecated `cocotb.result.TestFailure` class. https://docs.cocotb.org/en/stable/library_reference.html#cocotb.result.TestFailure has the details on what to use instead.

```
MODULE=coverage_test_main TESTCASE= TOPLEVEL=sample_module TOPLEVEL_LANG=verilog \
         /usr/bin/vvp -M /opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp 
     -.--ns INFO     gpi                                ..mbed/gpi_embed.cpp:76   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     gpi                                ../gpi/GpiCommon.cpp:101  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb                             Running on Icarus Verilog version 11.0 (stable)
     0.00ns INFO     cocotb                             Running tests with cocotb v2.0.0.dev0 from /opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb
     0.00ns INFO     cocotb                             Seeding Python random module with 1698972390
     0.00ns CRITICAL cocotb.regression                  Failed to import module coverage_test_main: cannot import name 'TestFailure' from 'cocotb.result' (/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/result.py)
     0.00ns INFO     cocotb.regression                  MODULE variable was "coverage_test_main"
     0.00ns INFO     cocotb.regression                  Traceback: 
     0.00ns INFO     cocotb.regression                  Traceback (most recent call last):
                                                          File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/regression.py", line 202, in _discover_tests
Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/__init__.py", line 210, in _initialise_testbench
    _initialise_testbench_(argv_)
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/__init__.py", line 293, in _initialise_testbench_
    regression_manager = RegressionManager.from_discovery(top)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/regression.py", line 173, in from_discovery
    return cls(dut, tests)
           ^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/regression.py", line 152, in __init__
    for test in tests:
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/regression.py", line 202, in _discover_tests
    module = _my_import(module_name)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/regression.py", line 71, in _my_import
    mod = __import__(name)
          ^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/_pytest/assertion/rewrite.py", line 186, in exec_module
    exec(co, module.__dict__)
  File "/home/runner/work/cocotb/cocotb/cocotb-coverage/tests/test_coverage/coverage_test_main.py", line 5, in <module>
    from cocotb.result import TestFailure
ImportError: cannot import name 'TestFailure' from 'cocotb.result' (/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/result.py)
Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/__init__.py", line 309, in _sim_event
    scheduler.log.error(msg)
    ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'log'
                                                            module = _my_import(module_name)
                                                                     ^^^^^^^^^^^^^^^^^^^^^^^
                                                          File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/regression.py", line 71, in _my_import
                                                            mod = __import__(name)
                                                                  ^^^^^^^^^^^^^^^^
                                                          File "/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/_pytest/assertion/rewrite.py", line 186, in exec_module
                                                            exec(co, module.__dict__)
                                                          File "/home/runner/work/cocotb/cocotb/cocotb-coverage/tests/test_coverage/coverage_test_main.py", line 5, in <module>
                                                            from cocotb.result import TestFailure
                                                        ImportError: cannot import name 'TestFailure' from 'cocotb.result' (/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/result.py)
                                                        
     0.00ns ERROR    gpi                                Passing event to upper layer failed
VCD info: dumpfile waveform.vcd opened for output.
ERROR: results.xml was not written by the simulation!
make[2]: *** [/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/share/makefiles/simulators/Makefile.icarus:96: results.xml] Error 1
make[2]: Leaving directory '/home/runner/work/cocotb/cocotb/cocotb-coverage/tests/test_coverage'
make[1]: *** [/opt/hostedtoolcache/Python/3.11.6/x64/lib/python3.11/site-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2
make[1]: Target 'all' not remade because of errors.
make[1]: Leaving directory '/home/runner/work/cocotb/cocotb/cocotb-coverage/tests/test_coverage'
make: *** [Makefile:3: all] Error 2
make: Leaving directory '/home/runner/work/cocotb/cocotb/cocotb-coverage/tests'
```

mciepluc, 2023-11-10:
I will fix this within a week

mciepluc, 2023-11-15:
@imphil fixed now

closed 2023-11-15 by mciepluc