mciepluc/cocotb-coverage #89: How to randomize a large address width ?

hurisson, 2024-01-14:
Hello. I encountered this problem when trying to randomize a 32-bit address and data. Randomization takes a long time, after which the test is barely simulated. According to the report, the occupied memory usage = 339997260 KB. Is there any other solution, without using list?

```
class apb_master_random_sequence(cocotb_coverage.crv.Randomized, uvm_sequence):
    def __init__(self, name, addr = None, data = None, direction = None, pstrb = None, pprot0 = None, pprot1 = None, pprot2 = None):
        cocotb_coverage.crv.Randomized.__init__(self)
        super(uvm_sequence, self).__init__("name")
        self.addr = data
        self.data = data
        self.add_rand("addr", list(range(0, 0xFFFF_FFFF)))
        self.add_rand("data", list(range(0, 0xFFFF_FFFF)))
....
```

mciepluc, 2024-01-14:
Hi @hurisson ,
You should not involve contraint solver for generating a simple random number. Please look at the tutorials, there is a section about performance issues: 
https://cocotb-coverage.readthedocs.io/en/latest/tutorials.html#randomization-order-and-performance-issues

hurisson, 2024-01-16:
@mciepluc  Thank you very much for your help

closed 2024-01-16 by hurisson